ra3xdh/qucs #4: Can't open Qucs schematic

ldpgh, 2016-10-06:
Schematic containing digital elemente can' t be shown ... loading content is stopped (no crash) and the error message is somehow misleading. It seems to be caused by missing inverter from "components/digital components" made with regular Qucs 0.0.19.

Schematic for test: [test_digsim_inv.sch.txt](https://github.com/ra3xdh/qucs/files/513780/test_digsim_inv.sch.txt)

ldpgh, 2016-10-07:
Now it's becoming mysterious ... have closed all running Qucs programs over night

Now ...
- Qucs complains about missing SpiceOptions ... might be OK
- Spice4Qucs complains too about missing SpiceOptions ... surprising
    - components/spiceOptions really not there 
    - but components/digital_components is back

Have hacked removed manually <SpiceOptions ....> from the file and it's working again.
What it controls is still not clear to me .... maybe more RTFM required

ldpgh, 2016-10-07:
It seems to depend on the simulator selection when Qucs is closed.

If Qucsator was previously active
- components/digital_components available
- controls/SPICE_specific_sections not available

If Ngspice was previously active
- components/digital_components not available
- controls/SPICE_specific_sections available

Can lead to a situation where its not possible to open a schematic.

ra3xdh, 2016-10-07:
It's not a bug. Qucs-S removes simulator-incompatible components (like Microwave devices for SPICE) from the components tree. And these components become unable to load. It was done to prevent mad simulator error message when trying to simulate circuit with incompatible components. And it was selected to prevent loading of such circuits. There exists the following PR  https://github.com/Qucs/qucs/pull/507 which enables loading of unknown components. Also it concerns components form future versions, etc. Unknown components will be replaced by placeholders. 

There may be alternative solution. We can hide components form component tree, but it will be available inside simulator and could be loaded. It requires a significant upgrade of Qucs components tree. It may lead to further divergence of Qucs and Qucs-S.

closed 2016-10-30 by ra3xdh