ra3xdh/qucs #7: Qucsator does not accept SweepModel property

haroldbai, 2016-10-09:
The Parameter sweep has SweepModel property which is set to "false". When running simulation, Qucsator triggers an error that SweepModel is invalid. The error message is shown below.

line 8: checker error, extraneous property `SweepModel' is invalid in `SW:SW1'
line 8: checker error, no such variable `false' used in a `SW:SW1' property
checker notice, variable `V1.I' in equation `Ic' not yet defined
checker notice, variable `Ib' in equation `Beta' not yet defined

I used Qucsator from Qucs 0.19RC1.

ra3xdh, 2016-10-10:
I have corrected Qucsator netlist entry generation. See https://github.com/ra3xdh/qucs/commit/bcb0d25543f9bcd3276c16862b08e35a277a2b10 for fix.

closed 2016-10-10 by ra3xdh

nanthony21, 2017-03-14:
I'm still getting this bug on ubuntu

ra3xdh, 2017-03-14:
What version/branch are you using? I have just tested the latest 0.0.19S release packages, and parameter sweep analysis was successfully performed with Qucsator. Please check that you are using the latest stable version https://ra3xdh.github.io/#inst