ra3xdh/qucs #8: Error while writing data to file ... data not available after simulation

ldpgh, 2016-10-17:
Issue root cause:
- schematic has a symbol with 2 different named pins. Both pins are connected to the **same** subcircuit-net.
- Top-level net `vtrl` and `vbp ` (shown below) are connected to those symbol pins with just a single net underneath.

---

Current implemenation: 
- one line statement
   - `write zt_vco_dc.txt VPr1#branch v(avdd) v(fout) v(ib) v(vbp) v(vctrl) v(ven)`
- It fails **after** the TRAN-analysis. The simulator throws an error message (see [ngspice#334](https://sourceforge.net/p/ngspice/bugs/334/)). The data are not exported to the result file. Just some kinde of a header is stored in the result file.

---

Requested implementation change:
- split the write statement **or**  just issue write statement without expressions to keep all
   - `.save VPr1#branch v(avdd) v(fout) v(ib) v(vbp) v(vctrl) v(ven) `
   - `     ... `
   - `write zt_vco_dc.txt`
- see [ngspice#334](https://sourceforge.net/p/ngspice/bugs/334/) for further details
   - save statement seems to be tolerant to invalid elements
   - write statement aborts if at least one invalid elements has been found

---

Side effect of the issue:
- The result file with just the header may (or may not) be the root cause of Qucs-S crash (?).
   - debugger complains about zero-pointer when converting raw-data to Qucs-data
   - no further debugging made for this topic
   - no longer seen after the schematic has been modified

ra3xdh, 2016-10-18:
Thank you for pointing this featurebug of Ngspice. I will consider your solution with dividing `write` into `save`+`write`. Another good solution may be to extend an existing netlist checker. For example, it checks for Ngspice reserved words as node names now. And it could be added a test for more than one subcircuit port connected to the same net.

ldpgh, 2016-10-18:
If the fix is a bigger action we need to talk about sweeping parameters.
Have a problem with parameters within SIN ( .... ). I'm still debugging Ngspice behavior.

Netlist checker ... ACK ... is important. But I think there are other more important issues.
Check for pin collissions ... No, it could be a feature. Maybe smart to keep the current state (and save the time) until a severe issue.

What's the right place for discussions?

ra3xdh, 2016-10-18:
Yes, you can continue discussion here. What are the exact problems with SIN() and Par.Sweep?

ldpgh, 2016-10-18:
I tried to sweep the FREQ element in SIN ( Vo ... Freq ....) from within Qucs.

Though the test case tries to sweep Vo ... same issue
 . . . see comments (OK, Error) in the attached netlist file [my_sweep_ix.cir.txt](https://github.com/ra3xdh/qucs/files/536857/my_sweep_ix.cir.txt)

But right now found memresistor (Ngspice-26/examples/memristor/memristor_x.sp
) example using one more coding style
 . . . V1 0 1 DC 0.1 ac 1 sin(0 'vmax' '1/stime')

If Qucs tries to support many simulators it's better to let the netlister (C, C++, python, ...) do the calculations and not rely on special command languages of the simulator. Unroll such sweeps is an easy task. Looks ugly (without idents) but less issues on the long term. 
In can go up to the point to start for each alter statement a single simulation call.
Qucs is collecting and tranlating the results data alreay.

ra3xdh, 2016-10-19:
I will think how to better implement vector parameters in Par.Sweep. I don't like the current way of `Parameter Sweep` implementation for Ngspice.  And I am planning its refactoring. Ngspice has no `.STEP` simulation. And it's needed to emulate it with auto-generated Nutmeg script. Nutmeg is very flexible and look like traditional programming language. But it's optimized for human code writing and reading. And machine-generated Nutmeg code may have several limitations. Your solution with unwrapping of all Par.Sweeps at netlist building stage may be good hack. I will consider it. Probably I will raise parameter sweep at netlister stage, if better solution will not be found. 

You can use `Nutmeg script` simulation to implement vector parameters sweep. This simulation allows to implement nonstandard simulations. Here is attached example:
[image]