ra3xdh/qucs #12: Netlist not OK for Shortened Disabled Components

gogobers, 2016-11-28:
If components are disabled by Short (i.e. the green cross through it) the netlist is not OK. For example a circuit containing only AC-source and 50Ohme resistor, relevant lines are (if both elements are active)
 R1 Vr 0  50
 V1 Vr 0 DC 0 SIN(0 1 1G 0 0) AC 1
collapsing to 
 RR10 Vr Vr 0RR11 Vr 0 0V1 Vr 0 DC 0 SIN(0 1 1G 0 0) AC 1
if the resistor is diabled. Of course this throws an error: 
 Error on line 2 : rr10 vr vr 0rr11 vr 0 0v1 vr 0 dc 0 sin(0 1 1g 0 0) ac 1
    unknown parameter (vr) 
Workaround is of course to really replace the disabled component by a wire.
This was for 0.0.19_RC8 with Spice support, selected simulator: ngspice, but same for XYCE.
Thanks Gerald

ra3xdh, 2016-11-29:
Yes, there was a bug. See https://github.com/ra3xdh/qucs/commit/ae1660113db3a8a57267b2d7e572a43d10b580e8 for fix. Now shortened components could be processed normally. This schematic is processed correctly:
[image]
Here is netlist for it:
~~~
* Qucs 0.0.19  /home/vvk/.qucs/11.sch
R1 0 _net0  200
RR20 _net0 _net1 0
V1 _net1 0 DC 1
R3 _net0 _net1  50
~~~

gogobers, 2016-11-29:
OK, Thanks!

closed 2016-11-29 by gogobers