ra3xdh/qucs #13: Harmonic Balance

gogobers, 2016-11-28:
In earlier Version of Spice4Qucs variant of QUCS there used to be a Harmonic Balance simulation, that worked with XYCE (quite nicely for me). I have read there was an issue with XYCE and they wanted to change something. Anyway, the HB-Simulations gone in RC8, Any chance to get it back, or a chance I can help?
Thanks
Gerald

ra3xdh, 2016-11-29:
It is not a bug. I haven't removed HB simulation. "Default simulator" setting was introduced a some time ago. Qucs hides simulator-incompatible components from the GUI now. You need to select XYCE as the default simulator and restart Qucs. HB will appear again. See this screenshot:
[image]

gogobers, 2016-11-29:
That makes my Day, Thank you!

closed 2016-11-29 by gogobers