ra3xdh/qucs #14: Qucs diodes work as conductors using xyce

Emmanuel-DUFOUR, 2017-01-07:
Qucs diodes work as conductors using xyce, they have to be replaced with "D-diodes"

ra3xdh, 2017-01-08:
Hello, Emmanuel. I cannot reproduce this bug. Please send me a schematic. You have probably switched accidentally switched diode to the "shorted" mode (highlighted by green cross). You can see at the attached screenshot that diodes are operational with XYCE. I used the latest code from `qucs-s-stable` branch. 
[image]

You can see at netlist that .MODEL is placed at the end of the netlist. It uses `D()` model:
~~~
* Qucs 0.0.19  /tmp/dio.sch
R1 Node1 _net0  200
VPr2 _net0 Node2 DC 0
V1 Node1 0 DC 0 SIN(0 0.8 1MEG 0 0) AC 0.8
D_1N4148_1 0 Node2 DMOD_D_1N4148_1 AREA=1.0 Temp=26.85
.MODEL DMOD_D_1N4148_1 D (LEVEL = 2 Is=2.22e-10 N=1.65 Cj0=4e-12 M=0.333 Vj=0.7 Fc=0.5 Rs=0.0686 Tt=5.76e-09 Ikf=0 Kf=0 Af=1 Bv=75 Ibv=1e-06 Xti=3 Eg=1.11 Tbv1=0 Trs1=0 Tnom=26.85 )
.tran 2e-08 4e-06 0 
.PRINT  tran format=raw file=dio_tran.txt I(VPr2) v(Node1) v(Node2) 
.END
~~~

Emmanuel-DUFOUR, 2017-01-08:
I use the 0.0.19S-rc8
this time I cant even launch the simulation on xyce serial
I get:
Netlist error: There was 1 undefined symbol in .PRINT command:
DIODE_TRAN.TXT

see attached schematic

ra3xdh, 2017-01-08:
Please send me this schematic privately, because Guthub drops attachments.

ra3xdh, 2017-01-08:
Diode is indeed shortened in this schematic. But it is shortened by user. Anode and cathode are connected to the same node. Of course, such diode will be (and should be) handled as a wire. And there is no bug here. If you remove one wire at this schematic diode will be simulated properly. Netlist also contains proper entry for diode. 

The reason of `undefined symbol in .PRINT command` error is different. XYCE cannot handle spaces in dataset file names. It is featurebug of XYCE. I will try to add a fix to bypass it. Temporary solution is only don't use spaces in schematic file names.

Emmanuel-DUFOUR, 2017-01-08:
Hi
I forgot to delete the short and remove the space from the filename
try with this attachement, I get 0.5 amps  current (!)

ra3xdh, 2017-01-08:
I cannot reproduce diode bug again. I tried Linux and WinXP in VirtualBox. Simulation always works properly. The current is 0.0034 A at 1 V source. You are doing something wrong, but I cannot figure out what.

Emmanuel-DUFOUR, 2017-01-08:
Please find attached my screen on xyce 6.6 serial and 0.0.19S-rc8

ra3xdh, 2017-01-08:
Github dropped an attachment again. I suspected the reason was in Xyce-6.6, because I used Xyce-6.5. I have checked again Xyce-6.6+Qucs-S-RC8 combination and all worked properly. I used Linux and WinXP platforms. Here is the screenshot:
[image]

ra3xdh, 2017-01-08:
@Emmanuel-DUFOUR , I have found your attachment. Simulation results seems to be correct. See #15 to track spaces in schematic filename bug.

closed 2017-01-08 by ra3xdh

Emmanuel-DUFOUR, 2017-01-08:
This is not correct, the diode should block all current, in your example
the current should be near 0 instead of 0.08 A.
In your circuit:
5v/50 Ohm= 0.1 Ampere. You get 0.08A (almost 0.1A), the diode behaves like
a conductor instead of blocking the current!

ra3xdh, 2017-01-08:
Oh, it is clearer now. Sorry, I forgot about reverse polarity when inspecting your circuit. Of course diode current should be around zero. But your example is incorrect, because your diode model has reverse breakdown voltage `BV=0`. Diode always goes into the breakdown at zero bias. If I set `BV=10` all works as expected. Ngspice gives the same results. And there no bugs neither in Qucs-S neither in XYCE.

Emmanuel-DUFOUR, 2017-01-08:
ok that clears the issue thanks