ra3xdh/qucs #19: Cannot plot subcircuit values (Xyce)

tvrusso, 2017-04-19:
I am exploring use of Qucs-s as a Xyce schematic capture tool.

In my schematic, a common-emitter amplifier subcircuit with an internal node "VE" is instantiated in a trivial driver schematic.  I am attempting to plot the VDB(OUT) and VDB(XSUB1:VE) values.

VDB(OUT) works fine.  VDB(XSUB1:VE) appears in the GUI for creating the plot, and all the data for this signal are in the .dat file produced by the run.  The GUI allows me to select this signal for inclusion in the graph.  But the graph shows up with "<invalid>" in the axis label, and no data is plotted.

Github is refusing to allow me to attach the zip file of the schematic.  I have posted it to [my personal web site](http://pages.swcp.com/~russo/CommonEmitter_subcircuited.ZIP).

ra3xdh, 2017-04-20:
Yes, it's a bug. Qucs plotting system cannot plot a variable containing colon `:` symbol. I will look how to overcome this limitation.

ra3xdh, 2017-04-21:
The colon symbol ':' is reserved for dataset name specification in Qucs plotting system. So, `dataset:var` means the variable `var` from file `dataset.dat` And this doesn't allow to plot variable names containing colon. The only way to overcome this limitation is to replace the colon symbol to some safe symbol (for example `_`). I can add this fix soon.

tvrusso, 2017-04-22:
Hmmm.  That's unfortunate, since the colon is used in subcircuit node specifiers in Xyce (and SPICE, though the exact syntax is very different).

Thanks for looking into it, though.

ra3xdh, 2017-04-22:
Look at https://github.com/ra3xdh/qucs/commit/ae067f9447c10d826371049367c18c3ce5c69721 for possible fix. It replaces the colon by `_` symbol.

tvrusso, 2017-04-22:
That works for the attached circuit and others I've tested.  I was initially concerned because the underscore character is legal in node names, but the way you've done this doesn't break plotting of nodes with underscores in their names.  Looks good to me.  Thanks!

closed 2017-04-23 by ra3xdh