ra3xdh/qucs #20: Cannot build qucs-s from git repo

tvrusso, 2017-04-19:
I am building Qucs-s from the posted source tarball at https://ra3xdh.github.io on my FreeBSD machine.  It builds just fine.

Since it appears that there is ongoing development of Qucs-s and that some of the changes being posted are specific to Xyce (the code I'm interested in using), I would like to track Qucs-s right out of GitHub.

But the code on the qucs-s-stable branch, even from the tag supposedly associated with the release tarball, will not build.  It appears that significant changes have been introduced between commit de213a695 (the 0.0.19S tag) and https://github.com/ra3xdh/qucs/releases/download/0.0.19S/qucs-0.0.19S.tar.gz, specifically in CMakeLists.txt files.

It'd be great if somewhere there were instructions for the right way to build non-released versions of qucs-s directly from the git repo.  The Xyce team is really interested in the qucs-s project, and it would be useful from our perspective to be able to see how qucs-s development improves the usefulness of the tool with Xyce.

If I try to build qucs-s-stable branch code, using exactly the same process that works with the tarball code, I get down to a link phase, where it fails trying to build the qucs executable.  Looks like parts of the qucs build that are explicitly disabled in the qucs-s release tarball are still trying to get built by the github version:
[ 56%] Linking CXX executable qucs
CMakeFiles/qucs.dir/module.cpp.o: In function `Module::registerModules()':
/users/russo/src/qucs/qucs/qucs/module.cpp:(.text+0x3eed): undefined reference to `bsim3v34nMOS::info(QString&, char*&, bool)'
/users/russo/src/qucs/qucs/qucs/module.cpp:(.text+0x3f2f): undefined reference to `bsim3v34pMOS::info(QString&, char*&, bool)'
/users/russo/src/qucs/qucs/qucs/module.cpp:(.text+0x3f71): undefined reference to `bsim4v30nMOS::info(QString&, char*&, bool)'
/users/russo/src/qucs/qucs/qucs/module.cpp:(.text+0x3fb3): undefined reference to `bsim4v30pMOS::info(QString&, char*&, bool)'
/users/russo/src/qucs/qucs/qucs/module.cpp:(.text+0x3ff5): undefined reference to `hicumL0V1p2::info(QString&, char*&, bool)'
/users/russo/src/qucs/qucs/qucs/module.cpp:(.text+0x4037): undefined reference to `hicumL0V1p2::info_pnp(QString&, char*&, bool)'
/users/russo/src/qucs/qucs/qucs/module.cpp:(.text+0x4079): undefined reference to `hicumL0V1p2g::info(QString&, char*&, bool)'
/users/russo/src/qucs/qucs/qucs/module.cpp:(.text+0x40bb): undefined reference to `hicumL0V1p2g::info_pnp(QString&, char*&, bool)'
/users/russo/src/qucs/qucs/qucs/module.cpp:(.text+0x40fd): undefined reference to `hicumL0V1p3::info(QString&, char*&, bool)'
/users/russo/src/qucs/qucs/qucs/module.cpp:(.text+0x413f): undefined reference to `hicumL0V1p3::info_pnp(QString&, char*&, bool)'
/users/russo/src/qucs/qucs/qucs/module.cpp:(.text+0x4181): undefined reference to `hicumL2V2p23::info(QString&, char*&, bool)'
/users/russo/src/qucs/qucs/qucs/module.cpp:(.text+0x41c3): undefined reference to `hicumL2V2p24::info(QString&, char*&, bool)'
/users/russo/src/qucs/qucs/qucs/module.cpp:(.text+0x4205): undefined reference to `hicumL2V2p31n::info(QString&, char*&, bool)'
/users/russo/src/qucs/qucs/qucs/module.cpp:(.text+0x4247): undefined reference to `HBT_X::info(QString&, char*&, bool)'
c++: error: linker command failed with exit code 1 (use -v to see invocation)
*** Error code 1

tvrusso, 2017-04-19:
I should note also that I attempted to build the spice4qucs_current branch as well.  This is lacking all the CMake files needed to build the way that qucs-s documentation says to build, and when I try an autoconf build I get other failures.   After a few attempts at trying an out-of-source build (separate build directory) I find it won't work because bootstrap runs "configure" in the qucs-doc source directory, which breaks out-of-source builds.  If I build directly in the source tree:

$ ./bootstrap
$ env QT_SELECT=qt4 ./configure --with-mkadms=/usr/local/bin/admsXml

then configure works fine.  But make fails:

Making all in qucs-core
gmake[1]: Entering directory '/users/russo/src/qucs/qucs-core'
gmake  all-recursive
gmake[2]: Entering directory '/users/russo/src/qucs/qucs-core'
Making all in src
gmake[3]: Entering directory '/users/russo/src/qucs/qucs-core/src'
gmake[3]: *** No rule to make target 'gperfapphash.cpp', needed by 'equation.cpp'.  Stop.
gmake[3]: Leaving directory '/users/russo/src/qucs/qucs-core/src'
gmake[2]: *** [Makefile:775: all-recursive] Error 1
gmake[2]: Leaving directory '/users/russo/src/qucs/qucs-core'
gmake[1]: *** [Makefile:662: all] Error 2
gmake[1]: Leaving directory '/users/russo/src/qucs/qucs-core'
gmake: *** [Makefile:363: all-recursive] Error 1

ra3xdh, 2017-04-20:
@tvrusso , It's nice to see interest from Xyce team to Qucs-S subproject. The Qucs and Qucs-S projects are diverging, but I am trying to keep the mergeability of Qucs-S and mainline Qucs. So, the build procedure and branching model of Qucs-S became weird at some point. Yes, it's a good idea to put contribution guide somewhere at Qucs-S webpage. 

Currently there are three main branches in my home Qucs repository:
* `qucs-s-stable` It is release candidate for the next Qucs-S release. This branch provides side-by-side installation of Qucs and Qucs-S. Main application executable is renamed to `qucs-s` instead of `qucs`. The same for other tools and locations in installation tree. Only CMake is supported You need to set `WITH_SPICE` key for CMake to build it properly. 
* `spice4qucs_current` It is development branch. Use this branch to target patches. It uses traditional Qucs build system. Main application executable name is `qucs`. It may override system Qucs installation. There exists some experimental branches forked form this branch. I merge `spice4qucs_current` into `qucs-s-stable` time to time.
* `master` is identical with mainline Qucs and doesn't contain spice4qucs additions. I keep it synchronized with mainline Qucs. I also merge `master` into `qucs-s-stable` time to time.

Release tarball is obtained from `qucs-s-stable` with some Bash script https://github.com/ra3xdh/qucs_scripts/blob/master/qucs-s-makepkg . This script removes Qucsator and microwave utilities from Qucs-S codebase. I distribute only Qucs-GUI and some utilities with Qucs-S package. 

My changes only concentrated on GUI side, and you need to rebuild GUI only to test Qucs-S. You may take Qucsator from usual Qucs installation. 
Here I will give a short build instructions:

* For `qucs-s-stable` you need to set `WITH_SPICE` cmake key. Autotools will not work. Perform the following commnads to build all:
```
git clone https://github.com/ra3xdh/qucs/ -b qucs-s-stable
cd qucs/qucs
mkdir build
cd build
cmake .. -DWITH_SPICE=ON  -DCMAKE_INSTALL_PREFIX=/some_path
make
```
Then install and launch `qucs-s`

* For `spice4qucs_current` you may use traditional Qucs build instructions. You may need to rebuild only Qucs-GUI. Autotools also may not work here, and it's recommended to use CMake:
```
git clone https://github.com/ra3xdh/qucs/ -b spice4qucs_current
cd qucs/qucs
mkdir build
cd build
cmake ..  -DCMAKE_INSTALL_PREFIX=/some_path
make
```
Then install and launch `qucs`

The `gperf` build issues probably related to https://github.com/Qucs/qucs/issues/295 I haven't changed anything in `qucs-core` and it should be identical with mainline Qucs. This issue probably depends on `gperf` version in FreeBSD. I met this issue time to time on my Slackware machine.

tvrusso, 2017-04-20:
I see, we need not to build at the top-level of the qucs checkout, but in the qucs/qucs directory.   That was my big mistake, I was looking in qucs, not qucs/qucs.  Following your directions above, I *AM* able to build qucs-s-stable.  I'll try out spice4qucs_current next.   

Thanks for the response.

tvrusso, 2017-04-20:
I am also able to build spice4qucs_current as you have instructed.  I'll close this issue.

closed 2017-04-20 by tvrusso