ra3xdh/qucs #21: Default simulator display at bottom of window shows wrong value

tvrusso, 2017-04-19:
Though I have Xyce (Serial) selected as default simulator, and this is in fact used when I run the simulation, Qucs-s displays "XYCE (Parallel)" in the lower right corner where it is meant to display the default simulator.
[image]

I should also note that since Xyce is not an acronym, the correct way to display its name is "Xyce" not "XYCE".

tvrusso, 2017-04-19:
Just looked at the code, and this is because:

    QString spicecompat::getDefaultSimulatorName()
    {
        QString sim_lbl;
        switch (QucsSettings.DefaultSimulator) {
        case spicecompat::simQucsator: sim_lbl = QObject::tr("Qucsator");
            break;
        case spicecompat::simNgspice: sim_lbl = QObject::tr("Ngspice");
            break;
        case spicecompat::simSpiceOpus: sim_lbl = QObject::tr("SpiceOpus");
            break;
        case spicecompat::simXycePar: sim_lbl = QObject::tr("XYCE (Serial)");
            break;
        case spicecompat::simXyceSer: sim_lbl = QObject::tr("XYCE (Parallel)");
            break;
        default: sim_lbl = QObject::tr("Qucsator");
            break;
        }
        return sim_lbl;
    }
    
has serial and parallel xyce label text interchanged, and should be:

    QString spicecompat::getDefaultSimulatorName()
    {
        QString sim_lbl;
        switch (QucsSettings.DefaultSimulator) {
        case spicecompat::simQucsator: sim_lbl = QObject::tr("Qucsator");
            break;
        case spicecompat::simNgspice: sim_lbl = QObject::tr("Ngspice");
            break;
        case spicecompat::simSpiceOpus: sim_lbl = QObject::tr("SpiceOpus");
            break;
        case spicecompat::simXyceSer: sim_lbl = QObject::tr("Xyce (Serial)");
            break;
        case spicecompat::simXycePar: sim_lbl = QObject::tr("Xyce (Parallel)");
            break;
        default: sim_lbl = QObject::tr("Qucsator");
            break;
        }
        return sim_lbl;
    }

ra3xdh, 2017-04-20:
Yes, I will correct this bug.

ra3xdh, 2017-04-20:
See https://github.com/ra3xdh/qucs/commit/660e24e5b87ac370f870f53b4d8e20d83cba0f81 for correction

tvrusso, 2017-04-20:
Looks good!  Thanks.

closed 2017-04-21 by ra3xdh