ra3xdh/qucs #23: SpiceLibComp saves full path to user libraries, making it difficult to share schematics

tvrusso, 2017-04-25:
If a user creates a set of User Libraries by putting them into ~/.qucs/user_lib and then uses them in schematics, Qucs-s saves the full path to the library file in the metadata for the part.

If one bundles up one's schematic and libraries and shares them with someone else, they don't display properly, even if that user has dropped the library files into the correct ~/.qucs/user_lib directory.  That's because the full path is no longer pointing at "~/.qucs/user_lib," but rather to "/home/<firstuser>/.qucs/user_lib." This can also bite a user who is running qucs-s on multiple machines with different home directory paths (/users/<myname> on one, /home/<myname> on another, for example).  In fact, that's how I realized this was a problem.  It's easy enough to fix by double clicking on the default symbol that shows up and fixing the path, or by hand-editing the .sch file, but for a large schematic with lots of user-library parts, it could become a problem.

It would be nice if qucs-s could recognize "~" as short-hand for "my home directory" and save this in the library path name in SpiceLibComp parts rather than the actual full path to that specific user's home directory on that specific machine.

ra3xdh, 2017-04-26:
Usual `LibComp` in mainline Qucs has the same behavior. If you paste `LibComp` from the user library, it will store the full path too. And the relative path is used for system libraries. It's not so easy to fix as it seems. Replacing the home path by `~` or `$HOME` will solve the problem only for Unix, but Windows will require `%USERPROFILE%`. The proper solution will be to introduce default library search paths and library priorities. It will allow to specify relative library path or even only library name. I will think how to proper fix this limitation.

ra3xdh, 2017-10-01:
This issue was moved to ra3xdh/qucs_s#4

closed 2017-10-01 by ra3xdh