ra3xdh/qucs #24: Voluminous Xyce simulator output can crash Qucs-s

tvrusso, 2017-04-25:
The slotProcessOutput method of qucs/qucs/qucs/extsimkernels/xyce.cpp saves all console output from Xyce into a QString "output."  All of the output of the run from beginning to end winds up stored in this one variable, so it can be displayed all at once when the run is complete.

This is normally OK for release builds of Xyce, but development builds are very verbose and can cause a huge amount of output in a long run.  When this happens, eventually qucs-s gets a memory allocation error and simply crashes with a Qt error:

    Warning: Qt has caught an exception thrown from an event handler. Throwing exceptions from an
    event handler is not supported in Qt. You must reimplement QApplication::notify() and catch all 
    exceptions there.

The code then gets an abort trap (on the one system where I've seen it).

That's because the memory allocation error happens inside the slot handler, as it tries to grow the "output" variable.

One possible solution would be to wrap the "output += s" line in a try/catch, and do something graceful (perhaps discarding older output by clearing the string?  Flushing the string to the dialog box and then clearing it?) rather than just relying on Qt's abort.  Or maybe periodically send the data to the dialog box and clear the string again, to avoid it growing so much to begin with?

This is obviously not going to be a problem for most simulators (or even Xyce with release versions), but we did encounter it when running a verbose development build on a long-running circuit.  Switching to a non-verbose version avoided it entirely.

ra3xdh, 2017-04-26:
Thank you for finding this bug. I haven't ever met simulator output huge enough to crash the Qucs-S GUI. Another interesting question is what is capacity of `QPlainTextEdit` that is used to store the simaltor output. Switching to asynchronous simulator output processing will be the best fix of this bug. It's need to divide the simulator output and send it to the TextEditor by little portions. I am also considering removing the `Simulate with external simulator` dialog and redirecting the simulator output to message docks in the next versions. This also may help with fixing this bug.

tvrusso, 2017-04-26:
It is unusual for normal builds of Xyce to produce this much output, but when we're developing we often enable the full verbosity --- which is really, really verbose.  I only stumbled on this bug by using a development build on a long-running simulation --- none of that verbose output was useful, and the "right" solution was to stop using a verbose build.  But it's a bug nevertheless, since it results in a graceless exit and core dump of the gui.

ra3xdh, 2017-10-01:
This issue was moved to ra3xdh/qucs_s#5

closed 2017-10-01 by ra3xdh