ra3xdh/qucs #26: Schematic duplicating doesn't change document settings

antage, 2017-06-25:
I tried to duplicate a schematic and I observe strange behavior. After simulation (with ngspice) I get a tab with plot and data from original schematic instead of plot and data from duplicated schematic.

The reason for this behavior is data set and data display filenames aren't changed after duplicating and reference to original schematic.

I think when we duplicate a schematic we should rename data set, data display and octave script filenames.

ra3xdh, 2017-06-25:
Hi, Anton! What method of schematic duplication are you using? Don't copy schematic file. Use `SaveAs` command to make a copy of schematic.

antage, 2017-06-25:
@ra3xdh 
I use context menu:

[image]

ra3xdh, 2017-06-25:
Yes, there is a problem with duplication in mainline Qucs package. It may not work as expected. See also https://github.com/Qucs/qucs/issues/633 for details. Use `SaveAs` for schematic duplication and to bypass this featurebug.

closed 2017-10-01 by ra3xdh